Dtcreation/Firmware-Molise-Artillery #6: [FR] Add MeatPack support

jgason, 2021-02-09:
Hi, 

It would be really cool if you could add the support for the MeatPack compression for Octoprint (https://github.com/scottmudge/OctoPrint-MeatPack#firmware-with-meatpack-support)

Seems like the nightly builds of Marlin has it already so if luck is on our side it would be easy to do

Dtcreation, 2021-02-09:
Hi,

I only use Marlin Official release, if Meatpack is the next release, il will be in Molise.

There are a lot of bugs in Marlin nightly builds

closed 2021-02-10 by Dtcreation

Dtcreation, 2021-02-22:
Hi,

You can now activate it on Molise !!

jgason, 2021-02-27:
Bonjour,

 

y-a-t’il une ligne a decommenter dans la compilation ou est-ce dedans par defaut ?

 

Bien á vous

 

Jean-Luc Gason