Dtcreation/Firmware-Molise-Artillery #24: Heated bed for homing

eiffel31, 2021-02-23:
Heating the bed before every homing is sometimes not practical at all.
Anoyances:
- delay
- hot surface

Could be better if the bed temperature could be selected (some preset values and adjustments buttons) at each homing order instead of automatic heating to 50°C.

Probe leveling for example has no interest into bed heading, extruder heating is more appropriate.

Thanks!

Dtcreation, 2021-02-23:
Hi,

This feature was in high demand.
I will study the feedback

SckyzO, 2021-02-24:
Suggest to add this in option in Molise. I use this feature because when the bed heats up, the surface moves a little. Which makes the BLtouch even more precise.

eiffel31, 2021-02-24:
Hi,
Right, when the bed heats up, it will move and the shape of the surface will change. I'll try to make things more clear:

The bed heating value has to be user configurable for:
- manual bed leveling
- automatic bed leveling
- homing for print

The nozzle heating value has to be user configurable for:
- probe offset calibration
- homing for print

This issue should be in TFT, not in firmware. But TFT is provided in firmware...

SckyzO, 2021-02-24:
Hi, 
Ah ok ... I understand :)

closed 2021-02-25 by Dtcreation