Dtcreation/Firmware-Molise-Artillery #27: Wrong printing speed

eiffel31, 2021-02-25:
The printing speed seems to be higher than expected.

Prevision from Prusaslicer: 30h

Filament was not melted enough, so I increased the temperature and slow down the printing speed to 90% most of the time and 80% on some parts, some slower tests. Total print time: 28h45'. According to the reduced print speed and initial prevision, I expected at least 35h of print time.

28h45' is the printed value, it is correct according to the wall clock.

Is Prusaslicer so pessimistic?

SckyzO, 2021-02-25:
The slicer only provides an estimate ... Not a molise bug.

SckyzO, 2021-02-26:
Just, for information, do you try same model, same settings on Cura, S3D and Prusa ? 3 Slicer, same settings, 3 différents time for the same model for me :-)

eiffel31, 2021-02-26:
Do you mean opening the same gcode file gives these 3 different times?

SckyzO, 2021-02-26:
Yes. you can try. Each Slicer works differently. 
And all they do is estimate the time. It is not something "stable". In addition, each Slicer has its own way of working (movement, optimization, etc.)

Dtcreation, 2021-02-28:
Hi,

Not a firmware issue, slicer issue

closed 2021-02-28 by Dtcreation