Dtcreation/Firmware-Molise-Artillery #28: Wrong filament quantity

eiffel31, 2021-02-25:
Estimated filament length from Prusaslicer: 303.59m
Displayed at printing end: 98.43m

The displayed value is clearly wrong.
As the extruder is calibrated in mm, this should not happen.

If it is computed from mm3 back to filament length and configuration uses 3mm instead of 1.75mm as filament diameter, then it can be close to explain the value. This is just a guess...

Note: observed on X1.

SckyzO, 2021-02-25:
Not a molise problem. 

Is your slicer which calculate this. 
Did you optimize E step ? Calibre extruder ?

Also, slicer just estimate quantity of filament to extrude. Like a car. You estimate 450 km with A full tank of fuel and you will browse by car 500 km 😎 and maybe you browse 400 ... 😱

eiffel31, 2021-02-26:
The extruder default calibration is perfect (not the default from Artilery, but the default you set in Molise). It is the same value as my own measure with less than 0.1% difference (so bellow the measure precision).

The slicer is correct. The final print is 800g (checked) and nearly a full roll was used. The estimate was a bit heavier but I printed really fast so some under-extrusion occurred, so everything is perfectly inline.

The printed value is wrong: below 100m is absolutely wrong because this is 300g.

The difference with cars is that with printers everything is calibrated. Calibration errors can result in a few % of errors.

Here this is a bug: if the system says the extruder used 98.53m, how could it use more than 250m in real life? The extruder does not generate many extra steps by itself. This is a bug, of course the hard part is to locate where it is...

PS: je prends le temps de faire des observations précises et je prends le temps de reporter les problèmes de manière détaillée. Ce n'est pas pour casser les pieds, c'est pour aider à améliorer. L'imprimante est sympa et ça fait plaisir de voir des Français apporter des améliorations :-)

Amicalement

SckyzO, 2021-02-26:
Ok j'avais mal lu ton soucis. Je vais regarder dans Marlin si je trouve qqch

eiffel31, 2021-02-26:
J'ai ouvert le gcode dans un autre viewer, il trouve exactement pareil. Normal: la longueur n'est pas une estimation, c'est la somme des avancées de l'extrudeur demandées tout au long du fichier gcode.

La calibration n'intervient pas (elle intervient sur la réalité du filament utilisé, pas sur ce que le programme pense avoir utilisé).

SckyzO, 2021-02-27:
Avec un écart aussi grand je ne comprend pas ... j'ai mes imprimantes prises jusqu'a jeudi soir, je regarderai le weekend du 6/7 si je vois un truc. La je trouve juste bizarre autant d'écart. Cela t'es arrivé souvent ou juste sur 1 print ?

eiffel31, 2021-02-27:
Je viens seulement de découvrir et d'installer Molise. La version 1.3.0 est très récente, je n'ai pas fait beaucoup de print de 900g avec...

PS : j'ai aussi remarqué des choses étranges avec le paramètre vitesse. Sur certains print il affiche parfois 0, puis l'affichage bascule sur fan 100% puis il revient sur speed et cette fois c'est 100%. Parfois en voulant changer, il ouvre l'écran en considérant que la demande c'est 10 au lieu de 100. Très étrange... Tu l'as déjà remarqué ? Je crée un nouveau bug pour ça ?

Dtcreation, 2021-02-28:
Bonjour,

Molise est basé sur Marlin Officiel et sur le firmware Bigtreetech Officiel pour le TFT. Les modifications et optimisations de Molise pour les imprimantes Artillery ne touchent pas au "coeur du code" de ces deux firmwares afin de conserver une cohérence et pouvoir faire les montée en version sans problème. Il est clair que les firmwares Marlin et Bigtreetech contiennent des bugs que je ne souhaite pas corriger au niveau de Molise mais qu'il faut corriger au niveau "source".

Je te propose de signaler ces bugs soit chez Marlin : https://github.com/MarlinFirmware/Marlin soit chez BTT pour le TFT (le bug de vitesse quand on manipule le tft est connu : https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware.

Ici nous traiterons les bugs liés directement à Molise et à la configuration spécifique Artillery (les 7 premières sections de Molise) et aux optimisations apportées (TouchMi, SKR1.4, gestion des LED...)

Merci pour tes remontées très précises et ton temps à expliquer tout ça.

closed 2021-03-04 by Dtcreation