Dtcreation/Firmware-Molise-Artillery #32: Feature request : Smart Sensor

jpreculeau, 2021-03-02:
Hello,

Is it possible to add an option for the BTT Smart Sensor (on the mainboad to use with octoprint) ?
Doc of the bidule : https://github.com/bigtreetech/smart-filament-detection-module/tree/master/manual

Thanks for the great work.

Dtcreation, 2021-03-09:
Bonjour,

Ce capteur a des réglages trop spécifiques qui rentrent en conflit avec le capteur d'origine, je ne peux donc pas le rajouter en standard 🙇‍♂️

closed 2021-03-09 by Dtcreation

jpreculeau, 2021-03-09:
Ok je comprends. du coup pour l'utiliser il vaut mieux que je m'abstienne d'activer le capteur dans les sections de configuration et directement faire les modifs dans marlin ou dois je faire quelque chose en plus s'il y a des spécificités avec les ajouts de ton firmware ?