Dtcreation/Firmware-Molise-Artillery #33: Feature Request : Neopixel

jpreculeau, 2021-03-02:
Hello,

I would use neopixel with the skr 1.4T motherboard.
Could you add an option to enable neopixel instead of stock led ? (I'm using bltouch directly connect directly on the motherboard. So the stock led connector is still available.

Dtcreation, 2021-03-09:
Bonjour,

J'ai rajouté l'option Neopixel_Perso (dans la branche Master pour l'instant) avec le connecteur Neopixel de la SKR, si tu veux utiliser le connecteur LED pour mettre une Neopixel il faut utiliser l'option #define TOUCH_MI_NEOPIXEL et la déplacer dans la section 7.

Je prépare un méthode plus simple pour la prochaine version

jpreculeau, 2021-03-09:
Oui l'idée est d'utiliser le port neopixel sur la skr et de recycler le connecteur LED RGB existant sur la carte de l'extrudeur. utiliser les fils RGB de la nappe existante évitera un câble en plus de celui du BLtouch (qui chez moi n'utilise justement pas la nappe. façon gueroloco)