Dtcreation/Firmware-Molise-Artillery #46: 1.3.0 à 1.3.1 plus possible de compiler, il manque quelque chose dans les fichiers sources !?

AndroSteph, 2021-03-11:
Bonjour à tous,

Après de nombreux tests, j'arrive à la constatation suivante. En partant d'un "Visual Studio Code" installé récemment, en ayant réussi à installer le plugins nécessaire (très bien expliqué sur la vidéo !) et en essayant de compiler la version 1.3.0, j'y arrive sans problème (je ne suis pas programmeur, hein). 

Si je refais cette étape, j'y arrive sans problème avec mon PC rebooté ou pas, depuis diffèrent noms de répertoires où j'ai les fichiers sources.

Maintenant si je prend la version 1.3.1, et dans exactement les mêmes conditions, il m'est totalement impossible de lancer la compilation.

Sur le bandeau du bas j'ai un "truc" qui tourne, continuellement, avec ce message "PlatformIO: Loading tasks..." et si je lance tout de même en appuyant sur le petit "V" (toujours dans le bandeau du bas) il apparaît sur la droite ... "command 'platformio-ide-build' not found"

Alors que j'ai réussi à compiler la version 1.3.0 quelques minutes juste avant.

Encore une fois je ne suis pas un programmeur, mais je pense avoir fait tous les tests systématiques avant de poster ce message.

Je reste à disposition ci-besoin ;-)

PS: je bricole pour l'instant, n'ayant pas encore reçu ma X1

Dtcreation, 2021-03-11:
Bonjour,

Il n'y a aucun soucis avec la compilation (j'en fait plus de 20 par jours)
Tu dois avoir un soucis quelque part sur ton vscode.
De plus, ta compilation ne te sort pas d'erreur, juste il ne veut pas lancer la compilation.

AndroSteph, 2021-03-11:
Bonjour,

Hum, mais comment comprendre que dans le répertoire 1.3.0 ... je compile et dans le répertoire 1.3.1 ... impossible. Tout ça avec le même Visual Studio Code ?

Bon, je vais laisser tomber ... car ma logique est dépassée par rapport aux essais que je viens de faire, avec reboot du PC, relance de VSCode, etc.

Dtcreation, 2021-03-11:
Impossible à dire sans être devant le PC. Essaye de supprimer le répertoire 1.3.1, de le re-télécharger

SckyzO, 2021-03-11:
@AndroSteph : Quel est le message d'erreur généré ?

AndroSteph, 2021-03-11:
@Sckyz0

Sur le bandeau du bas j'ai un "truc" qui tourne, continuellement, avec ce message "**PlatformIO: Loading tasks...**" et si je lance tout de même en appuyant sur le petit "V" (toujours dans le bandeau du bas) il apparaît sur la droite ... "**command 'platformio-ide-build' not found**"

@Dtcreation 
J'ai téléchargé à nouveau les deux fichiers le .zip et le .gz ... décompilé les deux (deux répertoires différents). Là je peux compiler les deux répertoires, ouf

Désolé de vous avoir em... pour rien, je n'avais pas pensé à une corruption "invisible" dans le fichier, vraiment DESOLE ... suis impatient de recevoir mon imprimante et faire un don.

Merci pour toutes vos aides :-)

Dtcreation, 2021-03-11:
Super, pas de soucis, l'informatique est une science presque exacte 😜😉

closed 2021-03-11 by Dtcreation