Dtcreation/Firmware-Molise-Artillery #50: comparison between signed and unsigned integer expressions

dedelamalice, 2021-03-14:
Un petit souci au niveau de la compilation ces lignes sont en jaune dans vs code 
ca n'empêche pas la compilation du firmware

comparison between signed and unsigned integer expressions [-Wsign-compare]

{
	"resource": "/d:/Deveveloper/Firmware-Molise/Firmware-Molise-Artillery-1.3.1/Marlin/src/module/probe.cpp",
	"owner": "cpp",
	"severity": 4,
	"message": "comparison between signed and unsigned integer expressions [-Wsign-compare]",
	"startLineNumber": 373,
	"startColumn": 44,
	"endLineNumber": 373,
	"endColumn": 44
}
[image]

Dtcreation, 2021-03-16:
Bonjour,

C'est pas un bug juste une expression que VScode ne comprend pas bien.

Aucune incidence

dedelamalice, 2021-03-16:
ok Merci