Dtcreation/Firmware-Molise-Artillery #59: MBL Start from 15 mm

Jonweb08, 2021-04-05:
Hello, 
I'm in love with such firmware, i use it starting from first version on my X1 stock, what i noticed is that i flash the last version 1.4 MBL stock and when i need to create  my first mesh he started  from 15 mm each point, this is not suitable because you have to decresa manually each point till arrive to 0 and start to use the babystep, in my opinion 0,35 mm as starting point is more than enough. Thanks in advance to take in care it

gathursteem, 2021-04-18:
I also subscribe to the view that 15mm is a bit to high.  I understand that is made originally on purpose to avoid a bed crash of the nozzle.  

However on artillery printers you are supposed to do a bed leveling using the 4 screws before you do the MBL.    When I do that before, usually I just have to press to reset button that brings the nozzle back to 0 then you increase or decrease with the other buttons.

Ergonomically it would make sense to set the starting point at 0.35 or roundabouts, because from mu expereince with 4 artillery printers thats where you would probably go as a tolerance level when using the MBL.  Now it would be a bummer for for people go would not care using the screws and go straight to MBL ... but then there are always people that would miss something.

Maybe a good compromise is to set the starting point as an option on TFT ?

Jonweb08, 2021-04-25:
> I also subscribe to the view that 15mm is a bit to high. I understand that is made originally on purpose to avoid a bed crash of the nozzle.
> 
> However on artillery printers you are supposed to do a bed leveling using the 4 screws before you do the MBL. When I do that before, usually I just have to press to reset button that brings the nozzle back to 0 then you increase or decrease with the other buttons.
> 
> Ergonomically it would make sense to set the starting point at 0.35 or roundabouts, because from my experience with 4 artillery printers that's where you would probably end up as a tolerance level when using the MBL. Now, it would be a bummer for people who would not care using the screws and go straight to MBL ... but then there are always people that would miss something.
> 
> Maybe a good compromise is to set the starting point as an option on TFT ?

Good Idea, but honestly I don't know if it is possible to improve. we will wait the feedback

Dtcreation, 2021-05-01:
Not possible for the moment to have the option on the tft

closed 2021-05-01 by Dtcreation