Dtcreation/Firmware-Molise-Artillery #74: Problem with original TFT and MKS SGEN L V2.0

mfjnext, 2021-04-11:
Dear Dtcreation,
First of all, thanks for the work done on this firmware
I changed the motherboard on my genius and following that, unable to operate the original screen, I have the following message: no printer attached.
I have compiled the firmware several times by changing, for example, the numbers of SERIAL_PORT thinking that it came from there, but nothing to do.
I do not know if the problem comes from the firmware of the motherboard or that of the screen.
Using firmware from kind3r, everything works.
Would there be a solution to the problem?
Thank you in advance

gathursteem, 2021-04-14:
Seems like motherboard is not connected to the screen.  Either the cable has loosened or the there is a software disconnection in the menus (it is used to upgrade firmware without having to disconnect the cable.  if it is indeed how you upgraded it then you should repeat the same process to reconnect the board)

mfjnext, 2021-04-14:
Thanks for the information 
I left this detail out, I thought this problem story was old history with the new motherboard
I will try by unplugging the screen

mfjnext, 2021-04-14:
I just tried, nothing to do, I even reverse the TX and RX cables in case
I also try with a BTT TFT V3.0 screen, I have the exact same message

gathursteem, 2021-04-14:
If the screen cannot find the motherboard, there is either a connection that is not being made (loose cables, power ?) or that the motherboard has died.  Those MKS Gen l motherboard die quite a lot actually.  I had one that blew a mosfett and almost set fire to the printer.  Best is to replace it with a SKR 1.4 but if you cannot, the just a standard MKS gen l should do.

gathursteem, 2021-04-14:
Just a thought; did you make the changes for your Sgen L from stock ?  Guide is here https://artillery.n3t.ro/32bit-mks-sgen-l-upgrade.html

mfjnext, 2021-04-15:
thank you for the link, I have the v2, on the site it is the v1 which is used, it is not exactly the same mainbord
if I use the firmware which is on this site by compiling it for the v2, I do not have this problem of the screen precisely
I even took a btt tft35 v3 to test, exactly the same problem, with molise no printer attached and with that of kind3r it works

mfjnext, 2021-04-16:
I have just compiled the firmware version 1.3.2 and 1.3.2.1, it is from this version that there is a problem 
version 1.3.2 works

mfjnext, 2021-04-16:
I just found the problem, if I activate FILAMENT_RUNOUT_SENSOR in section 6, the screen no longer works
[image]

mfjnext, 2021-04-17:
problem solved
the following configuration must be used for the mks sgen l v2.0
#define FIL_RUNOUT_PIN P1_28

closed 2021-04-17 by mfjnext