Dtcreation/Firmware-Molise-Artillery #78: Error:Line Number is not Last Line Number+1, Last Line: xxxx

Poke255, 2021-04-20:
Bonjour,

Pendant les impressions via octoprint, le firmware envoie des messages d'erreur de type "Error:Line Number is not Last Line Number+1, Last Line: xxxx" accompagnés de beeps... le seule moyen que j'ai trouvé pour le moment est de déconnecter l'écran via le menu connexion pour arrêter l'affichage des messages   :-/
j'avais bien fait un M502 puis M500 juste après l'installation du firmware 1.4.1
Il n'y a pas de problème lors de l'impression via la clé usb.
Please Help ^^

mfjnext, 2021-04-28:
bonjour,
il faut déplacer le connecteur de l'écran sur le port EXP1 de la carte mère pour éviter les conflits avec octoprint

closed 2021-05-01 by Dtcreation

Poke255, 2021-05-02:
Merci ça fonctionne, je n'ai plus les messages d'erreur.