Dtcreation/Firmware-Molise-Artillery #83: SKR1.4 T + TMC2209 = "No E stepper plug left for Z2!"

ewaca, 2021-05-04:
on dirait qu'il m’attribue 2 hotends et que du coup il n'y a plus de place pour le Z2. sauf que je vois qu'une hotends
```
#define TEMP_SENSOR_0 1

#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
```

```

In file included from c:\skr1-4\marlin\src\inc\marlinconfig.h:32,
                 from c:\skr1-4\marlin\src\core\serial.h:24,
                 from Marlin\src\HAL\LPC1768\DebugMonitor.cpp:25:
c:\skr1-4\marlin\src\pins\pins.h:1370:8: error: #error "No E stepper plug left for Z2!"
 1370 |       #error "No E stepper plug left for Z2!"
      |        ^~~~~
In file included from c:\skr1-4\marlin\src\inc\marlinconfig.h:41,
                 from c:\skr1-4\marlin\src\core\serial.h:24,
                 from Marlin\src\HAL\LPC1768\DebugMonitor.cpp:25:
c:\skr1-4\marlin\src\inc\SanityCheck.h:610:4: error: #error "If NUM_Z_STEPPER_DRIVERS is 2, you must define stepper pins for Z2."
  610 |   #error "If NUM_Z_STEPPER_DRIVERS is 2, you must define stepper pins for Z2."
      |    ^~~~~
c:\skr1-4\marlin\src\inc\SanityCheck.h:1673:6: error: #error "TEMP_SENSOR_1 is required with 2 or more HOTENDS."
 1673 |     #error "TEMP_SENSOR_1 is required with 2 or more HOTENDS."
      |      ^~~~~
In file included from c:\skr1-4\marlin\src\inc\marlinconfig.h:32,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
c:\skr1-4\marlin\src\pins\pins.h:1370:8: error: #error "No E stepper plug left for Z2!"
 1370 |       #error "No E stepper plug left for Z2!"
      |        ^~~~~
In file included from c:\skr1-4\marlin\src\inc\marlinconfig.h:41,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
c:\skr1-4\marlin\src\inc\SanityCheck.h:610:4: error: #error "If NUM_Z_STEPPER_DRIVERS is 2, you must define stepper pins for Z2."
  610 |   #error "If NUM_Z_STEPPER_DRIVERS is 2, you must define stepper pins for Z2."
      |    ^~~~~
c:\skr1-4\marlin\src\inc\SanityCheck.h:1673:6: error: #error "TEMP_SENSOR_1 is required with 2 or more HOTENDS."
 1673 |     #error "TEMP_SENSOR_1 is required with 2 or more HOTENDS."
      |      ^~~~~
In file included from c:\skr1-4\marlin\src\inc\marlinconfig.h:32,
                 from Marlin\src\HAL\LPC1768\HAL_SPI.cpp:51:
c:\skr1-4\marlin\src\pins\pins.h:1370:8: error: #error "No E stepper plug left for Z2!"
 1370 |       #error "No E stepper plug left for Z2!"
      |        ^~~~~
In file included from c:\skr1-4\marlin\src\inc\marlinconfig.h:32,
                 from Marlin\src\HAL\LPC1768\eeprom_wired.cpp:24:
c:\skr1-4\marlin\src\pins\pins.h:1370:8: error: #error "No E stepper plug left for Z2!"
 1370 |       #error "No E stepper plug left for Z2!"
      |        ^~~~~
In file included from c:\skr1-4\marlin\src\inc\marlinconfig.h:32,
                 from Marlin\src\HAL\LPC1768\eeprom_sdcard.cpp:25:
c:\skr1-4\marlin\src\pins\pins.h:1370:8: error: #error "No E stepper plug left for Z2!"
 1370 |       #error "No E stepper plug left for Z2!"
      |        ^~~~~
In file included from c:\skr1-4\marlin\src\inc\marlinconfig.h:32,
                 from Marlin\src\HAL\LPC1768\eeprom_flash.cpp:39:
c:\skr1-4\marlin\src\pins\pins.h:1370:8: error: #error "No E stepper plug left for Z2!"
 1370 |       #error "No E stepper plug left for Z2!"
      |        ^~~~~
In file included from c:\skr1-4\marlin\src\inc\marlinconfig.h:41,
                 from Marlin\src\HAL\LPC1768\HAL_SPI.cpp:51:
c:\skr1-4\marlin\src\inc\SanityCheck.h:610:4: error: #error "If NUM_Z_STEPPER_DRIVERS is 2, you must define stepper pins for Z2."
  610 |   #error "If NUM_Z_STEPPER_DRIVERS is 2, you must define stepper pins for Z2."
      |    ^~~~~
In file included from c:\skr1-4\marlin\src\inc\marlinconfig.h:41,
                 from Marlin\src\HAL\LPC1768\eeprom_wired.cpp:24:
c:\skr1-4\marlin\src\inc\SanityCheck.h:610:4: error: #error "If NUM_Z_STEPPER_DRIVERS is 2, you must define stepper pins for Z2."
  610 |   #error "If NUM_Z_STEPPER_DRIVERS is 2, you must define stepper pins for Z2."
      |    ^~~~~
c:\skr1-4\marlin\src\inc\SanityCheck.h:1673:6: error: #error "TEMP_SENSOR_1 is required with 2 or more HOTENDS."
 1673 |     #error "TEMP_SENSOR_1 is required with 2 or more HOTENDS."
      |      ^~~~~
In file included from c:\skr1-4\marlin\src\inc\marlinconfig.h:32,
                 from c:\skr1-4\marlin\src\sd\cardreader.h:24,
                 from Marlin\src\HAL\LPC1768\main.cpp:38:
c:\skr1-4\marlin\src\pins\pins.h:1370:8: error: #error "No E stepper plug left for Z2!"
 1370 |       #error "No E stepper plug left for Z2!"
      |        ^~~~~
c:\skr1-4\marlin\src\inc\SanityCheck.h:1673:6: error: #error "TEMP_SENSOR_1 is required with 2 or more HOTENDS."
 1673 |     #error "TEMP_SENSOR_1 is required with 2 or more HOTENDS."
      |      ^~~~~
In file included from c:\skr1-4\marlin\src\inc\marlinconfig.h:41,
                 from Marlin\src\HAL\LPC1768\eeprom_sdcard.cpp:25:
c:\skr1-4\marlin\src\inc\SanityCheck.h:610:4: error: #error "If NUM_Z_STEPPER_DRIVERS is 2, you must define stepper pins for Z2."
  610 |   #error "If NUM_Z_STEPPER_DRIVERS is 2, you must define stepper pins for Z2."
      |    ^~~~~
c:\skr1-4\marlin\src\inc\SanityCheck.h:1673:6: error: #error "TEMP_SENSOR_1 is required with 2 or more HOTENDS."
 1673 |     #error "TEMP_SENSOR_1 is required with 2 or more HOTENDS."
      |      ^~~~~
In file included from c:\skr1-4\marlin\src\inc\marlinconfig.h:41,
                 from Marlin\src\HAL\LPC1768\eeprom_flash.cpp:39:
c:\skr1-4\marlin\src\inc\SanityCheck.h:610:4: error: #error "If NUM_Z_STEPPER_DRIVERS is 2, you must define stepper pins for Z2."
  610 |   #error "If NUM_Z_STEPPER_DRIVERS is 2, you must define stepper pins for Z2."
      |    ^~~~~
c:\skr1-4\marlin\src\inc\SanityCheck.h:1673:6: error: #error "TEMP_SENSOR_1 is required with 2 or more HOTENDS."
 1673 |     #error "TEMP_SENSOR_1 is required with 2 or more HOTENDS."
      |      ^~~~~
In file included from c:\skr1-4\marlin\src\inc\marlinconfig.h:41,
                 from c:\skr1-4\marlin\src\sd\cardreader.h:24,
                 from Marlin\src\HAL\LPC1768\main.cpp:38:
c:\skr1-4\marlin\src\inc\SanityCheck.h:610:4: error: #error "If NUM_Z_STEPPER_DRIVERS is 2, you must define stepper pins for Z2."
  610 |   #error "If NUM_Z_STEPPER_DRIVERS is 2, you must define stepper pins for Z2."
      |    ^~~~~
In file included from c:\skr1-4\marlin\src\inc\marlinconfig.h:32,
                 from Marlin\src\HAL\LPC1768\tft\tft_spi.cpp:23:
c:\skr1-4\marlin\src\pins\pins.h:1370:8: error: #error "No E stepper plug left for Z2!"
 1370 |       #error "No E stepper plug left for Z2!"
      |        ^~~~~
c:\skr1-4\marlin\src\inc\SanityCheck.h:1673:6: error: #error "TEMP_SENSOR_1 is required with 2 or more HOTENDS."
 1673 |     #error "TEMP_SENSOR_1 is required with 2 or more HOTENDS."
      |      ^~~~~
In file included from c:\skr1-4\marlin\src\inc\marlinconfig.h:41,
                 from Marlin\src\HAL\LPC1768\tft\tft_spi.cpp:23:
c:\skr1-4\marlin\src\inc\SanityCheck.h:610:4: error: #error "If NUM_Z_STEPPER_DRIVERS is 2, you must define stepper pins for Z2."
  610 |   #error "If NUM_Z_STEPPER_DRIVERS is 2, you must define stepper pins for Z2."
      |    ^~~~~
c:\skr1-4\marlin\src\inc\SanityCheck.h:1673:6: error: #error "TEMP_SENSOR_1 is required with 2 or more HOTENDS."
 1673 |     #error "TEMP_SENSOR_1 is required with 2 or more HOTENDS."
      |      ^~~~~
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\DebugMonitor.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_wired.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\tft\tft_spi.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\main.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_flash.cpp.o] Error 1
=========================================================================== [FAILED] Took 5.30 seconds ===========================================================================

Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:00:05.300

```

Dtcreation, 2021-05-05:
Il ne faut toucher qu'aux 7 premières sections du configuration.h

Merci de joindre vos fichiers de conf

ewaca, 2021-05-05:
Je n'ai justement touché qu'aux 7 premières sections. 
Au début, ayant un 2e en enstop en z, j'ai fait les modif nécessaires, et au moment de compiler j'avais cette erreur, du coup je suis reparti à 0 en ne decommentant que ce qu'il faut dans les 7 sections et j'ai toujours cette erreur.
Je posterais les 2 fichiers de configuration ce soir en rentrant.

ewaca, 2021-05-05:
j'ai tout repris depuis le début, j'ai extrait l'archive github et lancé VS code, modifié les 7 sections, et j'ai toujours la meme erreur. Voici les fichiers configs. Niveau Hardware j'ai une X1, avec une skr1.4 turbo, des TMC2209 en UART, le reste est stock (pas de bl touch, extrudeur de base ...)
[config.zip](https://github.com/Dtcreation/Firmware-Molise-Artillery/files/6429083/config.zip)

closed 2021-05-14 by ewaca