Dtcreation/Firmware-Molise-Artillery #93: [BUG] Z-align, unknown command

Zazou49, 2021-09-07:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When I want to do a Z-align, I've an error : `Unknown command : G34`

[image]

### Version of Marlin Firmware

Molise 2.3 (Marlin 2.0.9.1)

### Printer model

Artillery Genius

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

Zazou49, 2021-09-07:
As I can see in your video, it disable be default.

Can you share your configuration.h file for Genius stock with BL Touch ? I'll try to compile myself !

Thank you again

SgtWest10, 2021-09-07:
Maybe it's not a but.
Can i help you ?

SgtWest10, 2021-09-07:
@Zazou49
Maybe it's not a bug.
Can i help you ?

Zazou49, 2021-09-07:
I compiled myself the firmware with Z Auto Align enable and it seems working !

SgtWest10, 2021-09-07:
@Zazou49
be carefull, you need to remove the belt from the Z axes.

Zazou49, 2021-09-07:
Yes, I removed the belt before the Z align !

SgtWest10, 2021-09-07:
What do you want to know about to compile yourself ?

Zazou49, 2021-09-07:
I was afraid to enable / disable some options, but it was relatively easy with the video.

SgtWest10, 2021-09-07:
Use visual studio code and platformio extension.
You edit the file Marlin\configuration.adv, uncomment your hardware configuration in line 87 to 182.
You edit the file platformio.ini, the line 16 "default_envs = change_value", replace "change_value" with your board.
For your board value, it's in line 108 to 117, in file Marlin\configuration.adv.
Finally build the firmaware.
If you don't have error during compile, it's Okay.

SgtWest10, 2021-09-07:
@Zazou49 
if you have an original screen, I advise you to change the connection on the board to the exp1 port. this will avoid opening the printer to flash the firmware, using the disconnected button in the screen menu.

Zazou49, 2021-09-07:
Yes stock srceen, Open the printer is not needed, just click on "disconnect" on the screen and I can flash.

SgtWest10, 2021-09-07:
thank you for your work
@Dtcreation

SgtWest10, 2021-09-07:
@Zazou49 
Okay. Good.
On my sidewinder x1, I made the modification to make it work !

JRRN, 2021-11-06:
@Zazou49 if it is resolved, close the issue, please!

closed 2022-01-08 by Dtcreation