Dtcreation/Firmware-Molise-Artillery #98: [BUG] Hotends neopixel led for TouchMi don't light

capitainekurck, 2021-10-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I can light the led through the menu or it light when i do a nozzle PID with GCode but it don't light and change colour  while normal print or while normal operation

### Expected behavior

I expect the led light and change color while nozzle heat or while printing 

### Actual behavior

it don't light

### Steps to Reproduce

just printing

### Version of Marlin Firmware

last molise firmware (motherboard and TFT)

### Printer model

Xwinder X1

### Your Slicer

Cura

Dtcreation, 2022-01-26:
hi,

Do you print from SD card or Octoprint ?

SckyzO, 2022-10-31:
Pas de réponse, je clôture

closed 2022-10-31 by SckyzO