Dtcreation/Firmware-Molise-Artillery #101: [FR] Support de l'Artillery X2

SckyzO, 2021-11-21:
### Describe the feature you want

Artillery à publié les sources Marlin 2.0.9.1 avec le support de la X2. 
Je pense qu'on peut merge la conf pour un support de la X2 par molise :) 

https://github.com/artillery3d/sidewinder-x2-firmware

SckyzO, 2021-11-21:
Encore plus simple, Marlin vient d'intégrer la X2 dans ses sources : 
https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Artillery/Sidewinder%20X2

Dtcreation, 2022-01-08:
Release en cours !

closed 2022-01-08 by Dtcreation