Dtcreation/Firmware-Molise-Artillery #115: [BUG][ASX1] Impossible de piloter les axes sur le fimware TouchMi

gilbsgilbs, 2022-02-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Bonjour,

Il semble qu'une régression a été introduite dans la dernière release ([3.0.3](https://github.com/Dtcreation/Firmware-Molise-Artillery/releases/tag/3.0.3)) du firmware pour TouchMi sur la Sidewinder X1. Lorsque je flash ce firmware, plus aucun des axes de l'imprimante ne fonctionne, que ce soit piloté via octoprint ou via le TFT. Si je downgrade sur la [3.0.1](https://github.com/Dtcreation/Firmware-Molise-Artillery/releases/tag/3.0.1) (la [3.0.2](https://github.com/Dtcreation/Firmware-Molise-Artillery/releases/tag/3.0.2) n'étant que pour le BLTouch), tout fonctionne correctement et je peux de nouveau piloter mes axes.

### Bug Timeline

3.0.3

### Expected behavior

Les axes devraient bouger.

### Actual behavior

Les axes ne bougent pas.

### Steps to Reproduce

1. Flasher [le firmware 3.0.3 Stock+TouchMi pour la X1](https://github.com/Dtcreation/Firmware-Molise-Artillery/releases/download/3.0.3/Hex.Molise.3_0_3.X1.Stock.+.TouchMi.zip)
2. Essayer de bouger n'importe quel axe depuis le TFT => Rien ne se passe
4. Essayer de bouger n'importe quel axe avec Octoprint => Rien ne se passe

### Version of Marlin Firmware

2.0.9.3

### Printer model

Sidewinder X1

### Add-ons

TouchMi

### Additional information & file uploads

Pour l'anecdote, je viens d'installer le TouchMi et j'ai bien cru que j'avais raté mes câblages, mais non il semble bien que c'est le firmware :sweat_smile: …

Par contre, plus étonnant, j'ai aussi recablé le TFT sur EXP1 tel qu'indiqué dans la section [Installation et configuration](https://github.com/Dtcreation/Firmware-Molise-Artillery#installation-et-configuration) et pourtant je suis toujours obligé de débrancher le TFT pour pouvoir flasher :man_shrugging: . À noter que l'option "Déconnecter" le TFT ne fonctionne pas non plus et n'a jamais fonctionné sur ma X1 pour flasher un nouveau firmware.

Un grand merci pour ce superbe backport pour la Sidewinder X1.

SckyzO, 2022-10-31:
Bonjour, 
Ton problème est il tjrs d'actualité ?

gilbsgilbs, 2022-10-31:
Bonjour @SckyzO, merci pour la réponse.

Oui, ce problème est toujours d'actualité et concerne seulement la version 3.0.3 du firmware TouchMi. Je n'ai pas encore essayé de construire le firmware moi-même pour voir si c'était juste un problème lors de la release fournie sur GitHub ou si c'était un problème de configuration plus profond.

Je tourne toujours sur la 3.0.1, et j'avais pour projet à moyen terme de forker et me rebaser sur la dernière version de Marlin.

SckyzO, 2022-11-03:
Hello. Je bosse sur la v4. Basé sur Marlin 2.1.1. 
Je regarde ça pour voir les diff entre 3.0.1 et 3.0.3 pour mettre à jour la V4. Je te tiens au courant je m'occupe de cette issue

SckyzO, 2022-11-07:
Hello. Alors j'ai regardé, et je ne vois pas de différence. As tu essayé toi même de compiler à partir des sources à tout hasard ?

gilbsgilbs, 2022-11-07:
Salut,

Effectivement, [le diff](https://github.com/Dtcreation/Firmware-Molise-Artillery/compare/3.0.1...3.0.3) ne semble pas indiquer de changements pertinents 🤔 .

J'ai essayé de compiler la 3.0.3 moi-même et je reproduis bien le problème. Je n'ai cependant pas encore essayé de recompiler la 3.0.1 pour voir si je reproduis aussi, mais je prédis que j'aurai le même problème. Je soupçonne soit un flag oublié lors de la compilation de la 3.0.3 (et que j'aurais oublié aussi lorsque j'ai compilé moi-même), ou bien que la 3.0.1 serait en fait une autre version (qui n'a pas été compilée depuis les sources de la 3.0.1). Ça demande un peu plus d'investigations.

Est-ce que master est dans un état utilisable en ce moment ? Je peux essayer de la compiler ce soir et voir si ça se passe mieux avec Marlin 2.1.1.

SckyzO, 2022-11-07:
Oui bien sur. J'ai release Molise 4.0 (marlin 2.1.1) hier soir, et la branche master est la dernière version en 2.1.1 aussi. 
Je veux bien un retour comme ça je pourrai officiellement release des versions précompilés.

gilbsgilbs, 2022-11-07:
OK, je viens de tester et le problème semblé corrigé sur master 🎉 .

Merci pour ce travail !

closed 2022-11-07 by gilbsgilbs