Dtcreation/Firmware-Molise-Artillery #117: [BUG] BL TOUCH MAL FUNCTION MKS_SGENL_V1

JRRN, 2022-03-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When try to do homming or start auto bed leveling the extruder go to center of the plate and start to do homming but when touch the plate don't stop and try to go down. 

Since I changed the board to MKS SGEN L 1.0 because the stock drivers, another bltouch and the board fused.

Video attached.

### Bug Timeline

Detected when change broken MKSGENL TO MKS_SGENL

### Expected behavior

Same functionality as broken MKSGENL

### Actual behavior

Z (Extruder) don't stop when bltouch detect the plate when homming or when start ABL.

### Version of Marlin Firmware

DT Creation Molise 3.0.3

### Printer model

Artillery Genius V1

### Electronics

MKS SGEN L 1.0 MKS TMC2209 BLTOUCH-Waggster the rest is stock

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Additional information & file uploads

[Marlin.zip](https://github.com/Dtcreation/Firmware-Molise-Artillery/files/8193388/Marlin.zip)

VIDEO: https://drive.google.com/file/d/1ikiUIOu7Amw6cRGQB8wAcVq5O6DNpKn5/view?usp=sharing

JRRN, 2022-03-07:
Without SENSORLESS and without all DIAG Jumpers : https://drive.google.com/file/d/1pJBGwK09aosYsIEknd0X0BHWjLr1ESkR/view?usp=sharing

Without SENSORLESS and with JUMPERS: https://drive.google.com/file/d/1ikiUIOu7Amw6cRGQB8wAcVq5O6DNpKn5/view?usp=sharing
With SENSORLESS and with JUMPERS: anterior.https://drive.google.com/file/d/1ikiUIOu7Amw6cRGQB8wAcVq5O6DNpKn5/view?usp=sharing
With SENSORLESS and without JUMPERS, it goes to the end like the previous video but does not go down, just like the movement but without lowering the extruder and the bltouch does not work (https://drive.google.com/file/d/1ikiUIOu7Amw6cRGQB8wAcVq5O6DNpKn5/view?usp=sharing)

SckyzO, 2022-10-31:
Hello, sorry for delay. do you still have the same issue ?

JRRN, 2022-10-31:
> Hello, sorry for delay. do you still have the same issue ?

Yes, I have to switch off the machine and move manually the extruder to the 0,0 position.

SckyzO, 2022-11-06:
Ok, so first, can you try to compil new version in master branch, and tell me if you still have this issue.

JRRN, 2022-11-19:
Ok, @SckyzO I'll try tomorrow Sunday and get back to you with comments.

JRRN, 2022-11-19:
@SckyzO  BLTouch don't detect the min position.

Error enabling bed leveling on start machine. Master code used.

Attached output log and configuration.

[output.log](https://github.com/Dtcreation/Firmware-Molise-Artillery/files/10046693/output.log)
[configguration.txt](https://github.com/Dtcreation/Firmware-Molise-Artillery/files/10046695/configguration.txt)

SckyzO, 2022-11-20:
Did you try to do an Auto bed leveling after install Molise ? 
When you install Molise, first reset eeprom and do an ABL

JRRN, 2022-11-20:
Yes, I have tried. But the Z goes down and doesn't stop, it hits the bed, even with the sensor activated. It seems that the signal that it has reached the bed is not interpreted by the firmware.

JRRN, 2022-11-20:
https://mega.nz/file/O85lER6C#y7jbJvl0cy6gvGJxjx0frIymqQIzNY_ORii8UpY3LiY
[Configuration.txt](https://github.com/Dtcreation/Firmware-Molise-Artillery/files/10049340/Configuration.txt)

JRRN, 2022-12-08:
@SckyzO Fixed I'm going to prepara the documentation but with my wired connections the problem is that some pins on MKS SGEN L changes:

[image]

closed 2022-12-08 by JRRN

JRRN, 2022-12-08:
[image]

Fixed Pins