Dtcreation/Firmware-Molise-Artillery #142: [BUG] Hauteur de Z incorrecte lors de l'impression

Pi3rrick, 2022-11-20:
### Did you test the latest Molise code ?

Yes, and the problem still exists.

### Bug Description

Même après le réglage du Z-Offset la buse se place a environ 10mm au dessus du plateau et affiche à l'écran la hauteur de couche attendue (Par exemple 0.2) alors qu'elle est réellement bien plus haute. 
Le bed levelling et le home fonctionnent même si j'ai pu observer des "probe failing" car la position initiale du bltouch était trop haute et ne descendait pas assez pour rentrer en contact avec le bed. (Problème aléatoire)  

### Bug Timeline

New bug

### Version of Molise or Marlin Firmware

Release 4.0

### Printer model

X1

### Electronics

SKR 2 Rev A

### Add-ons

TMC 2209 + Tete Hemera

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Black-Widow, 2022-11-23:
Bonjour, j'ai le meme soucis

SckyzO, 2022-12-18:
Bonjour, 

Désolé de la réponse tardive j'étais pris sur d'autres projets. 

As tu résolu ton problème ?

closed 2023-01-06 by SckyzO