Dtcreation/Firmware-Molise-Artillery #146: [FR] (feature summary)

Darkwho58, 2023-04-04:
### Describe the feature you want

Bonjour, 

je souhaiterais modifier les dimensions de la x1 dans le codage. 
j’ai installé des rails linéaires ainsi qu’un touchmi (ce qui m’a fait déplacé mais endstop). 

A chaque fois que je lance un autobed leveling les chariots vienne en butée et force. 

Merci de votre réponse.

SckyzO, 2023-04-04:
Hello, 
Tes rails viennent en butée et force ? Ca veut dire que ton endstop ne les arrête pas ? Tu aurais un vidéo que je comprenne bien ?

Darkwho58, 2023-04-04:
Hello
Tiens la vidéo. 
merci pour ton retour

Darkwho58, 2023-04-04:
https://user-images.githubusercontent.com/72864046/229873030-fe54d12f-5978-4757-9c19-cdf3109834cf.MOV

SckyzO, 2023-04-05:
Ok alors oui peut être on peut réduire un peu le plateau, mais normalement c'est sensé s'arréter ... Tu avais bien pris le FW TouchMi ? 

Tu peux recompiler le Firmware en changeant les valeurs du Bed pour la X1/X2 : 

https://github.com/Dtcreation/Firmware-Molise-Artillery/blob/master/Marlin/Configuration.h#L2161

```
#if ENABLED(X1) || ENABLED(SWD_X2)
  #define X_BED_SIZE 305
  #define Y_BED_SIZE 305
#endif
```

par 

```
#if ENABLED(X1) || ENABLED(SWD_X2)
  #define X_BED_SIZE 300
  #define Y_BED_SIZE 300
#endif
```
Voir moins ...

Darkwho58, 2023-04-05:
Merci 

oui, j’ai pris celui à configurer soit même pour la neo pixel

Darkwho58, 2023-04-07:
Après énormément de bug, la x1 fonctionne mais … 
J’ai remarqué que les bugs venez du FW pour le TFT , j’ai installé celui de « bigtr » et cela fonctionne.

SckyzO, 2023-04-08:
Ha ... Bon alors je vais mettre à jour car je t'avoue que j'ai utilisé l'ancienne version que n'a pas été mis à jour depuis un moment.

Darkwho58, 2023-04-08:
Peut-être faire une version spécialement dédiée au touchmi 🙄